jeremiah-c-leary/vhdl-style-guide #1458: Rules port_011 and port_025 should not be used on component declaration

RasmusGOlsen, 2025-05-23:
**Environment**
Ubuntu 22.04
Python 3.13.2
vsg 3.31.0

**Describe the bug**
The ports in the entity and component declarations are checked using the naming rules port_011 or port_025. Only the ports in the entity declaration should be checked, as the port naming in the component declaration must match the declaration in the VHDL entity or Verilog module it describes.

**To Reproduce**
Check the following code snippet with either rule port_011 or port_025 enabled.
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity a is
  port (
    clk_i : in  std_logic;
    d_i   : in  std_logic;
    q_o   : out std_logic
  );
end entity a;

architecture structure of a is

  component b is
    port (
      c  : in  std_logic;
      d  : in  std_logic;
      q  : out std_logic
    );
  end component b;

begin

  u0: component b
    port map (
      c => clk_i,
      d => d_i,
      q => q_o
  );

end architecture structure;
```

**Expected behavior**
Port naming in component declaration must be ignored, because the naming of the ports is declared elsewhere.

jeremiah-c-leary, 2025-07-11:
Evening @RasmusGOlsen ,

I would agree those rules should not apply to component declarations.

I have pushed an update to the `issue-1458` branch.  When you get a chance could you check it out on your end and let me know if it fixes the issue?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-08-02:
Afternoon @RasmusGOlsen ,

Just wanted to ping you to see if you were able to verify the update is working.

Thanks,

--Jeremy

RasmusGOlsen, 2025-10-19:
I'm sorry, I haven't responded. I will take a look.

RasmusGOlsen, 2025-10-20:
Hi @jeremiah-c-leary, branch issue-1458 seems to solve the problem.