jeremiah-c-leary/vhdl-style-guide #1479: Is there a way to specify a regex for architecture_025 names?

ekigwana, 2025-08-01:
We are restricting architecture names and enforcing architecture name specification at instantiation.

All was going well until we encountered a case where one entity can has multiple architectures an one must be selected. The issue is that it is not possible to have same architecture name for two different architectures for one entity. In following with the spirit of this rule we'd like the ability to specify prefixes to names.

```
  architecture_025:
    disable : false
    names :
      - rtl*
      - empty
      - behave*
```

Is there a way to achieve the above?

ekigwana, 2025-08-27:
HI @JHertz5,

That you so much for you contribution. I have tested your solution and verified that it solves our issue. I can report that the regex operation gives us the required flexibility. I could not use regex expressions with square brackets e.g. ```behave_\[a-z_]*```. We do not have a need to bar use of non-alphabet characters.

Overall, this solution absolutely works for us! Thank you again.

ekigwana, 2025-08-27:
@JHertz5 ,

You are correct that it works as requested. I had a number in the architecture name to make sure that it would fail and did not realize that that was what I wanted. I tried it again and do confirm it works.

Thanks, I am looking forward to get this addition merged so that we can remove the exemptions.

jeremiah-c-leary, 2025-09-03:
Morning @ekigwana and @JHertz,

@JHertz5, thank you for adding this feature.  I will get this merged to master.

--Jeremy

closed 2025-09-03 by jeremiah-c-leary